EBiCS/EBiCS_Firmware #3: Status of branches

Koxx3, 2020-12-31:
Hello,
we are working on a new opensource firmware for Xiaomi escooter controller.
it has 3 shunts, external AOPS and a STM32F103C8T6.
It seems very close from your project and Lishui controller.
We would like to know the current status of your different branches ;)
Do you have a Telegram account ?
Regards and happy holidays ;)

Koxx3, 2020-12-31:
also, in the main branch, the project doesn't seems to be aligned with the CubeMX definition. as soon a I regenerate the configuration, some pins disappear and TIM3 is not usable ;)

jonpry, 2021-01-01:
Hi Koxx3, 

I am working on porting to the same Xiaomi controller. My branch is here: https://github.com/jonpry/EBiCS_Firmware
I have gotten as far as getting the motor to spin, both open loop and sort of in closed loop. The startup stutters in closed loop and seems to run at max speed. I am also running with the motor having 2 ohm resistors on each phase to avoid explosions. 

Some things I don't understand:
PB4, which is one of the Hall sensor inputs, appears to have a very strong pullup, or something else connected to it. Easy fix was to short the resistor on Hall A input, else it will not pull PB4 down low enough to trigger. It's a mystery, but problem for later. 

There are defines JSQR_PHASE_* at the top of the file. I'm not 100% on the order, changing which A,B,C corresponds to which channel 3,4,5 is important to get right. Working on that now

In the FOC code, I do not see anything that sets up a phase angle between the rotor and field, so how does it work at all?

Koxx3, 2021-01-02:
Hi @jonpry , we started a fresh new project based on STM32 Motor controller workbench.
If you are interested to participate, come with use ;)
https://t.me/joinchat/HDBKjVIwfiDAxQD5

stancecoke, 2021-01-05:
Sorry for writing late, I cleaned up the code today, so the most recent version is the Master now. I'm happy to see your interest in my project! :-)

closed 2021-01-08 by stancecoke