EBiCS/EBiCS_Firmware #8: Error compile

flumex2, 2021-09-01:
Hello,

when compile generate error:

Building target: LishuiFOC_01.elf
Invoking: MCU GCC Linker
arm-none-eabi-gcc -mcpu=cortex-m3 -mthumb -mfloat-abi=soft -L ../Drivers/CMSIS -
specs=nosys.specs -specs=nano.specs -T"../STM32F103C6Tx_FLASH.ld" -Wl,-Map=outpu
t.map -Wl,--gc-sections -o "LishuiFOC_01.elf" @"objects.list"  -larm_cortexM3l_m
ath -lm
c:/stm32/gcc-arm-none-eabi-10.3-2021.07/bin/../lib/gcc/arm-none-eabi/10.3.1/../.
./../../arm-none-eabi/bin/ld.exe: Src/display_kunteng.o:C:\Users\g\Desktop\ok pe
r luca\EBiCS_Firmware-master\EBiCS_Firmware-master\Debug/../Src/display_kunteng.
c:33: multiple definition of `huart1'; Src/display_ebics.o:C:\Users\g\Desktop\ok
 per luca\EBiCS_Firmware-master\EBiCS_Firmware-master\Debug/../Src/display_ebics
.c:13: first defined here
c:/stm32/gcc-arm-none-eabi-10.3-2021.07/bin/../lib/gcc/arm-none-eabi/10.3.1/../.
./../../arm-none-eabi/bin/ld.exe: Src/display_kunteng.o:C:\Users\g\Desktop\ok pe
r luca\EBiCS_Firmware-master\EBiCS_Firmware-master\Debug/../Src/display_kunteng.
c:21: multiple definition of `ui8_rx_buffer'; Src/display_ebics.o:C:\Users\g\Des
ktop\ok per luca\EBiCS_Firmware-master\EBiCS_Firmware-master\Debug/../Src/displa
y_ebics.c:14: first defined here
c:/stm32/gcc-arm-none-eabi-10.3-2021.07/bin/../lib/gcc/arm-none-eabi/10.3.1/../.
./../../arm-none-eabi/bin/ld.exe: Src/display_kunteng.o:C:\Users\g\Desktop\ok pe
r luca\EBiCS_Firmware-master\EBiCS_Firmware-master\Debug/../Src/display_kunteng.
c:13: multiple definition of `ui8_tx_buffer'; Src/display_ebics.o:C:\Users\g\Des
ktop\ok per luca\EBiCS_Firmware-master\EBiCS_Firmware-master\Debug/../Src/displa
y_ebics.c:15: first defined here
c:/stm32/gcc-arm-none-eabi-10.3-2021.07/bin/../lib/gcc/arm-none-eabi/10.3.1/../.
./../../arm-none-eabi/bin/ld.exe: Src/main.o:C:\Users\g\Desktop\ok per luca\EBiC
S_Firmware-master\EBiCS_Firmware-master\Debug/../Src/main.c:89: multiple definit
ion of `huart1'; Src/display_ebics.o:C:\Users\g\Desktop\ok per luca\EBiCS_Firmwa
re-master\EBiCS_Firmware-master\Debug/../Src/display_ebics.c:13: first defined h
ere
c:/stm32/gcc-arm-none-eabi-10.3-2021.07/bin/../lib/gcc/arm-none-eabi/10.3.1/../.
./../../arm-none-eabi/bin/ld.exe: Src/main.o:C:\Users\g\Desktop\ok per luca\EBiC
S_Firmware-master\EBiCS_Firmware-master\Debug/../Src/main.c:84: multiple definit
ion of `htim1'; Src/FOC.o:C:\Users\g\Desktop\ok per luca\EBiCS_Firmware-master\E
BiCS_Firmware-master\Debug/../Src/FOC.c:42: first defined here
collect2.exe: error: ld returned 1 exit status
make: *** [LishuiFOC_01.elf] Error 1

stancecoke, 2021-09-28:
Please post the content of your config.h!

stancecoke, 2022-02-03:
No reaction from opener.

closed 2022-02-03 by stancecoke