EBiCS/EBiCS_Firmware #12: Does this work with Rad Power Bikes?

larryqiann, 2022-10-18:
Hi,

I'm wondering whether this controller firmware is applicable to Rad Power Bikes with the Lishui controller, and I'm wondering whether the original display would be supported. If not, would the "open source eBike display" from EBiCS work with the protocol that this firmware uses?

Thanks!

erik-smit, 2022-12-08:
I got some spare parts from a RadRhino and I'm trying this firmware on it.
The PCB says: LSW12G-150Z-FOC-160801.

It has flashed, but it's not working great so far. It's burned 2 FETs and I'm still trying to figure out why.
The autodetect works... a little. But twice now after the autodetect one of the FETs has died and shorted inside.

Then after desoldering it, I boot it again, and when I try to run the motor on the throttle it works very stuttery.

erik-smit, 2022-12-09:
@stancecoke Should this firmware work on a LSW12G-150Z-FOC-160801?

I put some photos on: https://photos.app.goo.gl/tpZzBANRNPcBx6jM8

stancecoke, 2022-12-09:
I don't know exactly that controller, but I guess it will work.
For discussion, please join the Pedelecforum.
https://www.pedelecforum.de/forum/index.php?threads/open-source-firmware-fuer-lishui-controller.61113/

closed 2022-12-09 by stancecoke

erik-smit, 2022-12-11:
@larryqiann fwiw, it's now working for my radrhino controller. you can follow my learning at https://www.pedelecforum.de/forum/index.php?threads/open-source-firmware-fuer-lishui-controller.61113/post-1971427